athom-tech/esp32-configs #3: Home Assistant upgrade 2024.2.0 breaks update

GelsemaP, 2024-02-21:
After upgrading to ESPHome 2024.2.0 the smartplug v3 cannot be updated anymore. This is due to a breaking change as mentioned in https://github.com/esphome/esphome/pull/6095

https://github.com/athom-tech/esp32-configs/blame/91a600029692b52a447a597e4332928b73e2b2dc/athom-smart-plug.yaml#L100

hellcry37, 2024-02-21:
up for this, same issue

GelsemaP, 2024-02-22:
see the fork i created. If you edit the device in ESPhome in Home Assistant, change the package to point to: github://GelsemaP/esp32-configsv3/athom-smart-plug.yaml it will work. I've set the updates to 5 seconds, which might be too quick btw.

sanderlv, 2024-02-24:
Curious about the "formal" best way to update this. As others are mentioning:
https://community.home-assistant.io/t/update-interval-invalid-option-on-cse7766-platform-since-2024-2-0-update/693339

sanderlv, 2024-03-15:
Hey @athom-tech Fix please :-)