athom-tech/esp32-configs #13: Wifi signal changed from rssi (dB) into  signal strength (%)

erkr, 2024-07-02:
Hi

I  updated my Athom plug from a december 2023 version to the latest 2024.6.4
To my frustration, the WIFI signal sensor changed from RSSI (dB) into signal strength (%).
How to revert? I can judge percentages. All WIFI signals are represented in RSSI (dB) for decades.

Best Eric

Roving-Ronin, 2024-07-11:
@erkr  

100% = Good.   0% = Bad. 
   0dB = Good. 99 dB = Bad.

Either way if you want to have dB sensor show add the following to your devices local yaml file:

```
  - platform: wifi_signal
    name: "WiFi Signal"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: diagnostic
    internal: false
```

If you don't want the percentage to show, then do the opposite for it.

```
    # Reports the WiFi signal strength in %
  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Strength"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: diagnostic
    device_class: ""
    internal: true
```

erkr, 2024-07-11:
Thanks! great news it can be fixed by changing the yaml. This is new territory for me. Will give it a try.

Roving-Ronin, 2024-07-11:
Don't forget to close the issue, once you've tested and got working.

erkr, 2024-07-12:
As sead,this is new. I must have missed something. In particular where to define these modifications.

I get this error changing my devices local yaml:
```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/athom-smart-plug-v3-5090e8.yaml...
Failed config

sensor.wifi_signal: [source /config/esphome/athom-smart-plug-v3-5090e8.yaml:20]
  platform: wifi_signal
  name: WiFi Signal
  
  ID wifi_signal_db redefined! Check sensor->1->id.
  id: wifi_signal_db
  update_interval: 60s
  entity_category: diagnostic
  internal: False
  disabled_by_default: False
  force_update: False
  unit_of_measurement: dBm
  accuracy_decimals: 0
  device_class: signal_strength
  state_class: measurement
```
This is the modified yaml:
```
substitutions:
  name: athom-smart-plug-v3-5090e8
  friendly_name: Athom Plug V3 5090e8
packages:
  Athom_Technology.Smart_Plug_V3: github://athom-tech/esp32-configs/athom-smart-plug.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: qNbgnVYTRw1SOSBvhNfnpA3h9pZOyWHe6toudvhDCDE=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: diagnostic
    internal: false

  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Strength"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: diagnostic
    internal: true
```

erkr, 2024-07-12:
I'm a step further! This compiles and restores the dB sensor. The percentage is still there:
```
sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    id: !extend wifi_signal_db
    update_interval: 60s
    entity_category: diagnostic
    internal: false

  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Strength"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: diagnostic
    internal: true
```

erkr, 2024-07-13:
@Roving-Ronin thanks for the help so far. Some more guidance how to disable the strength in percentage sensor would be appreciated.
I close this issue though

closed 2024-07-13 by erkr

Roving-Ronin, 2024-07-14:
@erkr by adding the 'internal: true' the sensor should still remain, but only within the ESP device itself. i.e. It will NOT be published upon the web interface of the ESP, nor will it be published to HomeAssistant as a sensor. 
This is what I personally do with the WiFi dB sensor, I don't want to see it so its' set to 'internal: true', whilst the WiFi % sensor (that uses the data from the dB sensor, thats visible only within the ESP) is internal: false and therefore visible on the web interface and HA.

Are you referring to seeing the % sensor on your web interface of the ESP, or HomeAssistant? For the former, refresh the browser / purge cache, for HA go into the device, select the % sensor, click on the settings (cog wheel top right) then hit delete.

erkr, 2024-07-14:
@Roving-Ronin 
Thanks for your reply. I'm definitely referring the Signal strength sensor in %. That sensor remained present both in HA (I can disable it, not delete) and in the device web interface:
[image]

I think it's related to where I changed these settings. I changed the local device yaml, that includes the plug config from GitHub. So in need to overrule the original definitions (not a replacement). That is why I had to add `!extend` to the `id` part of the dB WiFi signal sensor.
The WiFi strength sensor in % is a copy without the id tag. That is probably why `internal: true` doesn't overrule the original definition. 
So my question is if there is a way to overrule this copy sensor as well.

This is my full local yaml:
``` 
substitutions:
  name: athom-smart-plug-v3-5090e8
  friendly_name: Athom Plug V3 5090e8
packages:
  Athom_Technology.Smart_Plug_V3: github://athom-tech/esp32-configs/athom-smart-plug.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: REDACTED

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    id: !extend wifi_signal_db
    update_interval: 60s
    entity_category: diagnostic
    internal: false

  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Strength"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: diagnostic
    internal: true
```

Roving-Ronin, 2024-07-14:
Try without the !extend, your not wanting to 'add on' to the config from GitHub, your wanting to overwrite / superceded what is from GitHub.

If that doesn't work, give this a try (update the name / friendly name to something that described what/where its used for. Power_Circuit is whatever 'Power' circuit you have in your switchboard, assuming you have a couple of 20A power circuits for power outlets [gpo] around the house:

```
substitutions:
  name: "gpo-clothes-dryer"
  friendly_name: "Clothes Dryer"
  device_description: "Athom Plug v3 - Phase B / GPO 3 - Clothes Dryer"
  # Allows ESP device to be automatically lined to an 'Area' in Home Assistant. Typically used for areas such as 'Lounge Room', 'Kitchen' etc
  location: "Laundry"
  # The phase in the home power supply, upon which this plug is utilsed. 1p = A. 3p = A, B or C.  (myHome GPO 1 = Phase C, GPO 2 = Phase A, GPO 3 = Phase B)
  power_circuit: "GPO 3"    

esphome:
  name_add_mac_suffix: false

api:
  encryption:
    key: REDACTED

packages:
  remote_package:
    url: https://github.com/Roving-Ronin/myHomeAssistant/
    ref: main 
    files: [esphome/athom-smart-plug-v3.yaml]
    refresh: 1d # optional

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    id: wifi_signal_db
    internal: false

  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Strength"
    internal: true
```

You might like some of the extra Energy sensors that are in this config (waiting for a PULL request to add them to the Athom repo configs for everyone to have).

erkr, 2024-07-14:
> Try without the !extend, your not wanting to 'add on' to the config from GitHub, your wanting to overwrite / superceded what is from GitHub.

Yes that is indeed what I want. But without `!extend` it doesn't compile (using the add on). 

> If that doesn't work, give this a try

I will do that later. But what I don't like is that I don't really grasp what I'm doing here. Basically only the packages part is different. But I will try if it compiles

Thanks !!

Roving-Ronin, 2024-07-14:
OK I see the dB sensor ID is clashing with the template one. I just tried it using !extend and validates correctly. I've updated the code to paste and that pulls off my enhanced template.  Gives you a dashboard like this (but with dB instead of %) and these sensors:

[image]

erkr, 2024-07-14:
Thanks for confirming the clash. As far I understand, the WiFi strength sensor doesn’t  clash as it is a copy without an own id. But modifications in the local device file remain  ignored. 

I guess if you try the other way around, so make the dB version internal false and the percentage internal true, you will end up having both as well

erkr, 2024-07-14:
@Roving-Ronin 

I think adding an ID to this sensor in the GitHub WiFi file would allow to modify the strength sensor as well:

``` 
   # Reports the WiFi signal strength in %
  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Signal"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: "diagnostic"
    device_class: ""
```

Roving-Ronin, 2024-07-14:
Try it by adding id: wifi_signal_percentage as a new line after the 'name' line, to a copy of that block of code in your local file.
If you the 'verify' it, you should find it has extended / merged that with GitHub code.

erkr, 2024-07-14:
I tried, compiles fine but these local settings for the copy sensor have no effect. 
I think the id needs to be added to that sensor in the referenced GitHub file, in order to enable the usage of !extend in the local file

Roving-Ronin, 2024-07-15:
I've just added 'id: wifi_strength' to the code, for the v3 plugs and its auto added into the existing PULL request that I have lodged already. 

IMHO I'd just fork the athom repository and make your own, then modify the file to remove the % sensor and change the dB sensor (just remove the line it has for 'internal: true') and then save it, doing so as your own 'BRANCH'.  Then just update the package in your local file to point it it instead, like:

```
substitutions:
  name: "gpo-clothes-dryer"
  friendly_name: "Clothes Dryer"
  device_description: "Athom Plug v3 - Phase B / GPO 3 - Clothes Dryer"
  # Allows ESP device to be automatically lined to an 'Area' in Home Assistant. Typically used for areas such as 'Lounge Room', 'Kitchen' etc
  location: "Laundry"
  # The phase in the home power supply, upon which this plug is utilsed. 1p = A. 3p = A, B or C.  (myHome GPO 1 = Phase C, GPO 2 = Phase A, GPO 3 = Phase B)
  power_circuit: "GPO 3"    

esphome:
  name_add_mac_suffix: false

api:
  encryption:
    key: REDACTED

packages:
  remote_package:
    url: https://github.com/erkr/esp32-configs/
    ref: NAME-OF-YOUR-BRANCH 
    files: [athom-smart-plug-v3.yaml]
    refresh: 1d # optional
```

erkr, 2024-07-15:
@Roving-Ronin 
I really appreciate you added the id to the official release!
I'm relatively new with GitHub and the processes around it and prefer to stick to the official version

I guess I now just have to wait till the id makes it to the official release. 

I now understand the extend part and will find my way. 

Thanks
Eric

erkr, 2024-07-17:
@Roving-Ronin 
Thanks for the suggestion and example of forking.
I managed to do that 👍