athom-tech/esp32-configs #56: Esphome 2025.7.x produces too large firmware sizes

deshark, 2025-07-17:
When arduino platform is used, firmware sizes exceed ota partition size. Switching to esp-idf helps.

Please update config files to use esp-idf.

wmrato, 2025-07-22:
**Still Not working for athom-smart-plug-v3
 Plus esp-idf. thows errors too.**

Error: The program size (2050152 bytes) is greater than maximum allowed (1835008 bytes)
RAM:   [==        ]  19.3% (used 63084 bytes from 327680 bytes)
Flash: [==========]  111.7% (used 2050152 bytes from 1835008 bytes)

💡 TIP: Your ESP32 with Arduino framework has run out of flash space.

To fix this, switch to the ESP-IDF framework which is more memory efficient:

1. In your YAML configuration, modify the framework section:

   esp32:
     framework:
       type: esp-idf

2. Clean build files and compile again

Note: ESP-IDF uses less flash space and provides better performance.
Some Arduino-specific libraries may need alternatives.

*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 252.84 seconds =========================

corrado-c, 2025-07-22:
> Svuota la cache e ricompila

yes I had already done the emptying and recompiling, now I'll try to empty again and also restart before compiling. 
but it still doesn't work

link in the program correct or did I change something by my mistake?

substitutions:
  name: athom-em-6-02a5e8
  friendly_name: Athom Energy Monitor 02a5e8

packages:
  Athom_Technology.Athom_Energy_Monitor(6_Channels): github://athom-tech/esp32-configs/athom-energy-monitor-x6.yaml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server: !remove

I also tried to remove the web server,

corrado-c, 2025-07-22:
Yes, I've already looked. I see hidden folders, but I don't have any hidden folders with that name or similar names, or anything else that might look like an athom file.
I tried inserting the entire program directly into my yaml file, avoiding the link to the package. That's the only way it works, not with the initial program, which refers to the link.

Thank you so much for the support and updates!!